faucamp/python-gsmmodem #44: Unhandled unsolicited modem notification: ['+CDS: 25',...

sah-anshu, 2015-01-09:
I am unable to see delivery status report. kindly advise. My logs are below:

DEBUG: 2015-01-09 17:45:55,829 write: AT+CVHU=0
DEBUG: 2015-01-09 17:45:55,914 response: ['OK']
DEBUG: 2015-01-09 17:46:16,187 notification: ['+CMTI: "SM",7']
DEBUG: 2015-01-09 17:46:16,188 SMS message received
DEBUG: 2015-01-09 17:46:16,190 write: AT+CPMS="SM"
DEBUG: 2015-01-09 17:46:16,289 response: ['+CPMS: 7,20,7,20,7,20', 'OK']
DEBUG: 2015-01-09 17:46:16,302 write: AT+CMGR=7
DEBUG: 2015-01-09 17:46:16,467 response: ['+CMGR: 0,,29', '0791198904100161240C911979099137640000511090718303220BC8B4421174CFD175D014', 'OK']
DEBUG: 2015-01-09 17:46:16,526 write: AT+CMGD=7,0
DEBUG: 2015-01-09 17:46:16,749 response: ['OK']
DEBUG: 2015-01-09 17:46:16,798 write: AT+CMGS=61
DEBUG: 2015-01-09 17:46:16,884 response: ['> ']
DEBUG: 2015-01-09 17:46:16,911 write: 0021000C91197909913764000036D3E614242F8FCB697B99AC038964B058AD058BB5603950ECA69BE17433D80A56D3CD603A1032AD5004DD73741D341501
DEBUG: 2015-01-09 17:46:31,352 response: ['+CMGS: 64', 'OK']
DEBUG: 2015-01-09 17:46:37,344 notification: ['+CDS: 25', '079119890400202306400C91197909913764511090718384225110907183252200']
DEBUG: 2015-01-09 17:46:37,345 Unhandled unsolicited modem notification: ['+CDS: 25', '079119890400202306400C91197909913764511090718384225110907183252200']

sah-anshu, 2015-01-09:
Can this be reason for the same ??? If yes what could be solution ?

DEBUG: 2015-01-09 17:38:11,714 write: AT+CNMI=2,1,0,2
DEBUG: 2015-01-09 17:38:11,803 response: ['+CMS ERROR: 604']
WARNING: 2015-01-09 17:38:11,828 Incoming SMS notifications not supported by modem. SMS receiving unavailable.

jthurner, 2015-01-09:
Your modem is set up to route the status reports directly (+CDS), you need to configure it to store the report and send the location to the TE (+CDSI). Check your modem manual and play around with AT+CNMI on the console.

closed 2015-01-10 by sah-anshu

luichanfe, 2015-06-05:
Hello, I have a modem that doesn't have the possibility to configure so it can store the report. It doesn't accept the command AT+CNMI=2,1,0,2, it only has AT+CNMI=2,1,0,0 and AT+CNMI=2,1,0,1.
How can I process the reports directly? (+CDS)

axwell, 2018-02-20:
did you manage to work with this ?