faucamp/python-gsmmodem #49: USC2 SMS not encoded correctly

obana1990, 2015-04-15:
Hi,

First of all thanks for all the hardwork, it's very useful. I was able to connect and send sms no problem. Then I tried to send unicode message like chinese and arabic text, it sends out successfully but the content delivered at the destination wasn't the original message instead it's just a bunch of weird symbols.

I'm running the script "sendsms.py" under tools directory. 
I'm using wavecom modem for your information.

Thanks in advance.

sah-anshu, 2015-04-17:
Please try sending in PDU mode.

obana1990, 2015-04-17:
I am sending on PDU mode. modem.textMode = False ?

sah-anshu, 2015-04-17:
Hi
Correct Statement is below:
    modem.smsTextMode = False 

Ref: https://github.com/faucamp/python-gsmmodem/blob/master/examples/sms_handler_demo.py

obana1990, 2015-04-17:
Hi, thank you so much. I will try it out after I got home tonight!

obana1990, 2015-04-18:
Hi,

I double check my code and yes I did modem.smsTextMode = False but it still giving wrong result.

MaplerStory, 2015-05-13:
You should use
`modem.sendSms("7036253126",u"哈哈")`
thus, if you use PDU mode, then you should send an unicode text