faucamp/python-gsmmodem #56: Exception in thread (python 2.7)/ Failed to parse PDU-mode SMS message

gOmp, 2015-09-07:
```
pi@raspberrypi ~/gsm_lab/external2/python-gsmmodem/examples $ sudo python sms_handler_demo.py
```
```
Initializing modem...
INFO: Connecting to modem on port /dev/ttyAMA0 at 19200bps
DEBUG: write: ATZ
DEBUG: response: ['ATZ\r', 'OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\rATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['AT+CFUN?\r', '+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['AT+CMEE=1\r', 'OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['AT+CPIN?\r', '+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['AT+CLAC\r', '+CLAC: ', 'AT#CLS', 'AT&C', 'AT&D', 'AT&F', 'AT&S', 'AT&V', 'AT&W', 'AT*CELLLOCK', 'AT*PSAHPLMN', 'AT*PSALS', 'AT*PSCCDN', 'AT*PSCELLLOCK', 'AT*PSCHRU', 'AT*PSCIPH', 'AT*PSCNAP', 'AT+*PSCPBS', 'AT*PSCPOF', 'AT*PSCSCN', 'AT*PSCSP', 'AT*PSCSSC', 'AT*PSDCIN', 'AT*PSFSNT', 'AT*PSGAAT', 'AT*PSHPLMN', 'AT*PSHZNT', 'AT*PSINFN', 'AT*PSLOCUP', 'AT*PSMBNB', 'AT*PSMEMCAP', 'AT*PSNTRG', 'AT*PSNWID', 'AT*PSOPNM', 'AT*PSPNNR', 'AT*PSPNNU', 'AT*PSPRAS', 'AT*PSRDBS', 'AT*PSRIC', 'AT*PSSEAV', 'AT*PSSREP', 'AT*PSSSURC', 'AT*PSSTK', 'AT*PSSTKI', 'AT*PSUTTZ', 'AT*PSVMWIR', 'AT*PSVMWN', 'AT*SLSHELL', 'ATA', 'ATA53', 'ATATT', 'ATAUDG', 'ATAUTEST', 'ATB', 'ATCAAS', 'AT+CACM', 'ATCADC', 'ATCADDFORBID', 'ATCAGC', 'AT+CALA', 'AT+CALD', 'AT+CALM', 'ATCALS', 'AT+CAMM', 'AT+CAOC', 'ATCAUTORUN', 'ATCBAND', 'ATCBC', 'ATCBRP', 'AT+CBST', 'ATCBTE', 'ATCBUZZERRING', 'ATCCALR', 'AT+CCFC', 'AT+CCID', 'AT+CCLK', 'ATCCODE', 'ATCCPD', 'ATCCVM', 'AT+CCWA', 'AT+CCWE', 'ATCDEVICE', 'ATCDNSCFG', 'ATCDNSGIP', 'ATCDRIND', 'ATCDSCB', 'ATCDTMFURC', 'AT+CEER', 'ATCELLFORBID', 'AT+CELLSWITCH', 'ATCEMNL', 'AT+CENG', 'ATCEXTBUT', 'ATCEXTERNTONE', 'ATCEXTHS', 'ATCFGRI', 'ATCFSDFILE', 'ATCFSGFIS', 'ATCFSGFRS', 'ATCFSINIT', 'ATCFSLIST', 'ATCFSREN', 'ATCFSRFILE', 'ATCFSTERM', 'ATCFSWFILE', 'AT+CFUN', 'AT+CGACT', 'ATCGAS', 'AT+CGATT', 'AT+CGCLASS', 'AT+CGCMOD', 'AT+CGCS', 'AT+CGDATA', 'AT+CGDCONT', 'AT+CGEQMIN', 'AT+CGEQREQ', 'AT+CGEREP', 'AT+CGID', 'AT+CGMI', 'AT+CGMM', 'AT+CGMR', 'ATCGMSCLASS', 'AT+CGPADDR', 'AT+CGQMIN', 'AT+CGQREQ', 'AT+CGREG', 'AT+CGSMS', 'AT+CGSN', 'ATCHDM', 'ATCHF', 'ATCHFA', 'AT+CHLD', 'AT+CHUP', 'ATCIFSR', 'ATCIICR', 'AT+CIMI', 'AT+CIND', 'ATCIODELTA', 'ATCIPACK', 'ATCIPATS', 'ATCIPCCFG', 'ATCIPCLOSE', 'ATCIPCSGP', 'ATCIPCTL', 'ATCIPDPDP', 'ATCIPFLT', 'ATCIPGSMLOC', 'ATCIPHEAD', 'ATCIPMODE', 'ATCIPMUX', 'ATCIPOPTION', 'ATCIPPING', 'ATCIPQRCLOSE', 'ATCIPQSEND', 'ATCIPRDTIMER', 'ATCIPRXGET', 'ATCIPRXISS', 'ATCIPSCONT', 'ATCIPSEND', 'ATCIPSENDBUF', 'ATCIPSENDHEX', 'ATCIPSERVER', 'ATCIPSGTXT', 'ATCIPSHOWTP', 'ATCIPSHUT', 'ATCIPSPRT', 'ATCIPSRIP', 'ATCIPSTART', 'ATCIPSTATUS', 'ATCIPSTTIMER', 'ATCIPTKA', 'ATCIPTXISS', 'ATCIPUDPMODE', 'AT+CIURC', 'AT+CLAC', 'AT+CLAN', 'AT+CLCC', 'AT+CLCK', 'ATCLDTMF', 'AT+CLIP', 'AT+CLIR', 'ATCLPORT', 'ATCLTS', 'AT+CLVL', 'AT+CMEC', 'AT+CMEE', 'AT+CMER', 'AT+CMGD', 'AT+CMGDA', 'AT+CMGF', 'ATCMGHEX', 'AT+CMGL', 'AT+CMGR', 'AT+CMGS', 'AT+CMGW', 'ATCMIC', 'AT+CMMS', 'ATCMMSBCC', 'ATCMMSCC', 'ATCMMSCID', 'ATCMMSCURL', 'ATCMMSDELBCC', 'ATCMMSDELCC', 'ATCMMSDELFILE', 'ATCMMSDELRECP', 'ATCMMSDOWN', 'ATCMMSEDIT', 'ATCMMSINIT', 'ATCMMSPROFILE', 'ATCMMSPROTO', 'ATCMMSRDPUSH', 'ATCMMSREAD', 'ATCMMSRECP', 'ATCMMSRECV', 'ATCMMSSCONT', 'ATCMMSSEND', 'ATCMMSSENDCFG', 'ATCMMSSTATUS', 'ATCMMSTERM', 'ATCMMSTIMEOUT', 'ATCMMSUA', 'ATCMMSVIEW', 'AT+CMOD', 'AT+CMSS', 'ATCMTE', 'AT+CMUT', 'AT+CMUX', 'ATCNETLIGHT', 'AT+CNETSCAN', 'ATCNITZNN', 'AT+CNMA', 'AT+CNMI', 'ATCNTP', 'ATCNTPCID', 'AT+CNUM', 'AT+COLP', 'AT+COPN', 'AT+COPS', 'ATCPAMR', 'AT+CPAS', 'AT+CPBF', 'AT+CPBR', 'AT+CPBS', 'AT+CPBW', 'AT+CPIN', 'AT+CPLS', 'AT+CPMS', 'AT+CPOL', 'AT+CPOWD', 'AT+CPRODFLAG', 'AT+CPUC', 'AT+CPWC', 'AT+CPWD', 'ATCQUICKPDP', 'AT+CR', 'ATCR99', 'AT+CRC', 'AT+CREG', 'AT+CRES', 'ATCRIPULSE', 'AT+CRLP', 'AT+CRMP', 'AT+CRNQP', 'AT+CRSL', 'AT+CRSM', 'AT+CRSN', 'ATCRTXVMA', 'AT+CSAS', 'AT+CSCA', 'AT+CSCB', 'ATCSCLK', 'AT+CSCS', 'AT+CSDH', 'ATCSDT', 'ATCSGS', 'AT+CSGT', 'AT+CSIM', 'AT+CSMINS', 'AT+CSMP', 'AT+CSMS', 'AT+CSNS', 'AT+CSPN', 'AT+CSQ', 'AT+CSSN', 'AT+CSTA', 'ATCSTT', 'ATCSUB', 'AT+CSVM', 'ATCTBURST', 'AT+CTFR', 'ATCTZR', 'ATCTZU', 'ATCUDHI', 'ATCUSACC', 'AT+CUSD', 'AT+CVHU', 'ATCWDTEN', 'ATCWHITELIST', 'ATD', 'ATD>', 'ATDDET', 'ATE', 'ATECHO', 'ATEMAILCID', 'ATEMAILSSL', 'ATEMAILTO', 'AT+EXUNSOL', 'AT+FCLASS', 'AT+FMI', 'AT+FMM', 'AT+FMR', 'AT+FRH', 'AT+FRM', 'AT+FTH', 'AT+FTM', 'ATFTPCID', 'ATFTPDELE', 'ATFTPETGET', 'ATFTPETPUT', 'ATFTPEXTGET', 'ATFTPEXTPUT', 'ATFTPFILEPUT', 'ATFTPGET', 'ATFTPGETNAME', 'ATFTPGETPATH', 'ATFTPLIST', 'ATFTPMKD', 'ATFTPMODE', 'ATFTPPORT', 'ATFTPPUT', 'ATFTPPUTNAME', 'ATFTPPUTOPT', 'ATFTPPUTPATH', 'ATFTPPW', 'ATFTPQUIT', 'ATFTPREST', 'ATFTPRMD', 'ATFTPSCONT', 'ATFTPSERV', 'ATFTPSIZE', 'ATFTPSMODE', 'ATFTPSTATE', 'ATFTPTIMEOUT', 'ATFTPTYPE', 'ATFTPUN', 'ATFTPV', 'AT+GCAP', 'AT+GMI', 'AT+GMM', 'AT+GMR', 'AT+GOI', 'AT+GPRSCELL', 'ATGSMBUSY', 'AT+GSN', 'ATGSV', 'ATH', 'ATHTTPACTION', 'ATHTTPDATA', 'ATHTTPINIT', 'ATHTTPPARA', 'ATHTTPREAD', 'ATHTTPSCONT', 'ATHTTPSSL', 'ATHTTPSTATUS', 'ATHTTPTERM', 'AT+HVOIC', 'ATI', 'AT+ICF', 'AT+IFC', 'AT+IPR', 'ATL', 'ATM', 'ATMORING', 'ATN', 'ATNETLOCK', 'ATO', 'ATP', 'AT+PACSP', 'ATPINTEST', 'ATPOP3CMD', 'ATPOP3DEL', 'ATPOP3IN', 'ATPOP3LIST', 'ATPOP3NUM', 'ATPOP3OUT', 'ATPOP3READ', 'ATPOP3RSET', 'ATPOP3SRV', 'ATPOP3UIDL', 'ATQ', 'ATS0', 'ATS1', 'ATS10', 'ATS11', 'ATS2', 'ATS3', 'ATS4', 'ATS5', 'ATS6', 'ATS7', 'ATS8', 'ATS95', 'ATSAPBR', 'ATSCLASS0', 'ATSGPIO', 'ATSIDET', 'ATSIMECHO', 'AT+SIMEI', 'ATSIMTONE', 'ATSJDR', 'ATSKPD', 'ATSLEDS', 'ATSMTPAUTH', 'ATSMTPBODY', 'ATSMTPCS', 'ATSMTPFILE', 'ATSMTPFROM', 'ATSMTPFT', 'ATSMTPRCPT', 'ATSMTPSEND', 'ATSMTPSRV', 'ATSMTPSUB', 'ATSPIC', 'ATSPWM', 'ATSRSPT', 'ATSSLCERT', 'ATSSLOPT', 'ATSTTONE', 'ATSVR', 'ATT', 'ATV', 'ATVIP', 'ATVR', 'AT+VTD', 'AT+VTS', 'ATW', 'ATX', 'ATZ', 'OK']
DEBUG: write: AT+CGMI
DEBUG: response: ['AT+CGMI\r', 'SIMCOM_Ltd', 'OK']
DEBUG: write: AT+ZPAS?
DEBUG: response: ['AT+ZPAS?\r', 'ERROR']
INFO: Unknown/generic modem type - will use polling for call state updates
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['AT+COPS=3,0\r', 'OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['AT+CMGF=0\r', 'OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['AT+CSCA?\r', '+CSCA: "+972586279099",145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['AT+CSMP=49,167,0,0\r', 'OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['AT+CSCA?\r', '+CSCA: "+972586279099",145', 'OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['AT+CPMS=?\r', '+CPMS: ("SM"),("SM"),("SM")', 'OK']
DEBUG: write: AT+CPMS="SM","SM","SM"
DEBUG: response: ['AT+CPMS="SM","SM","SM"\r', '+CPMS: 12,100,12,100,12,100', 'OK']
DEBUG: write: AT+CNMI=2,1,0,2
DEBUG: response: ['AT+CNMI=2,1,0,2\r', '+CMS ERROR: 303']
WARNING: Incoming SMS notifications not supported by modem. SMS receiving unavailable.
DEBUG: write: AT+CLIP=1
DEBUG: response: ['AT+CLIP=1\r', 'OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['AT+CRC=1\r', 'OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['AT+CVHU=0\r', 'OK']
Waiting for SMS message...
```
```
DEBUG: notification: ['+CMTI: "SM",13']
DEBUG: SMS message received
DEBUG: write: AT+CPMS="SM"
DEBUG: response: ['AT+CPMS="SM"\r', '+CPMS: 13,100,13,100,13,100', 'OK']
DEBUG: write: AT+CMGR=13
DEBUG: response: ['AT+CMGR=13\r', '+CMGR: 0,"",25', '0791795268721940040C9179528421334100005190705152912106F6B03BCD0E03', 'OK']
```
```
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 552, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 505, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/python_gsmmodem-0.9-py2.7.egg/gsmmodem/modem.py", line 869, in __threadedHandleModemNotification
    self._handleSmsReceived(line)
  File "/usr/local/lib/python2.7/dist-packages/python_gsmmodem-0.9-py2.7.egg/gsmmodem/modem.py", line 998, in _handleSmsReceived
    sms = self.readStoredSms(msgIndex, msgMemory)
  File "/usr/local/lib/python2.7/dist-packages/python_gsmmodem-0.9-py2.7.egg/gsmmodem/modem.py", line 1059, in readStoredSms
    raise CommandError('Failed to parse PDU-mode SMS message +CMGR response: {0}'.format(msgData))
CommandError: Failed to parse PDU-mode SMS message +CMGR response: ['AT+CMGR=13\r', '+CMGR: 0,"",25', '0791795268721940040C9179528421334100005190705152912106F6B03BCD0E03', 'OK']
```

darryllane, 2016-11-01:
Im getting a similar issue with the same manufacturer (SIMCOM Ltd). Did you have any luck with this? I can only assume the config differs for the modem. Are you using the sim900 or sim908?

gOmp, 2016-11-02:
@darryllane , I've used sim900.
sorry, had no luck with it, so I've used another approach for my project.
GL

darryllane, 2016-11-02:
Yes i think i will be taking a different approach. Did you go down pyserial
route?