faucamp/python-gsmmodem #60: Incoming SMS doesn't work on SIM800L

kkroo, 2015-11-17:
Executing the command `AT+CNMI=2,1,0,2` fails because the fourth argument *<ds>* is not 0 or 1.

Changing to  `AT+CNMI=2,1,0,1` fixes it 

https://www.adafruit.com/datasheets/sim800_series_at_command_manual_v1.01.pdf

kkroo, 2015-11-17:
This particular command isn't commented so I am not sure what the fourth argument is and how changing it to 1 will affect backward compatibility

bennyslbs, 2016-08-08:
The ETSI spec p. 41-42 says the fourth argument is <ds>s:
http://www.etsi.org/deliver/etsi_ts/127000_127099/127005/13.00.00_60/ts_127005v130000p.pdf

The SIM800L
supports those modes:
OK
at+cnmi=?
+CNMI: (0-3),(0-3),(0,2),(0,1),(0,1)

OK
atz
OK
at+cnmi?
+CNMI: 2,1,0,0,0

OK

The default CNMI value (2,1,0,0,0) works fine for me, but delivery reports are not delivered.
To get the delivery reports I have changed to 'AT+CNMI=2,1,0,1,0'. Unfortunately the last one gives that the status reports are delivered as a +CDS response, not a CDSI response.
This gives that the format is: '+CDS <length>', 'PDU message'

Maybe try 'AT+CNMI=2,1,0,2,0', and if it don't work then 'AT+CNMI=2,1,0,1,0' and keep in mind that parsing status notifications isn't supported ... or the smart way: check output from at+cnmi=?