faucamp/python-gsmmodem #62: Incorrect SIM card PIN entered for PIN-unlocked SIM

Jancs-E, 2015-11-21:
Hi!
module gives mentioned error if used with GSM modem (ID 05c6:6000 Qualcomm, Inc. Siemens SG75) having PIN-unlocked SIM-card.
I tried to enter the historical one the SIM had before unlocking as sendsms.py and identify-modem.py can not be used without giving PIN (without modification it seems).
The modem works well with chat (from ppp package) and gammu (although the stability is poor).
Janis

Jancs-E, 2015-11-21:
if I give the following command: ./sendsms.py -i /dev/ttyUSB0 -p none -d NUMBER
I get:
Connecting to GSM modem on /dev/ttyUSB0...
Traceback (most recent call last):
  File "./sendsms.py", line 88, in <module>
    main()
  File "./sendsms.py", line 53, in main
    modem.connect(args.pin)
  File "/usr/lib/python2.7/site-packages/gsmmodem/modem.py", line 198, in connect
    self._unlockSim(pin)
  File "/usr/lib/python2.7/site-packages/gsmmodem/modem.py", line 368, in _unlockSim
    self.write('AT+CPIN="{0}"'.format(pin))
  File "/usr/lib/python2.7/site-packages/gsmmodem/modem.py", line 428, in write
    raise CommandError(data)
gsmmodem.exceptions.CommandError: AT+CPIN="none"

In the same time AT+CPIN? replies with +CPIN:READY//OK

Jancs-E, 2015-11-21:
I found that removing the space on line 366 in modem.py so that the test looks so: 
if self.write('AT+CPIN?')[0] != '+CPIN:READY':
solves half of the problem (I think the both versions '+CPIN:READY' and '+CPIN: READY' should be tested for compatibility or space to be stripped before test). 
The next is to solve issue with SMSC number which is either not retrieved or misinterpreted.

Jancs-E, 2015-11-29:
Hi!
I created the patch solving SMSC issue by adding the option to specify it and solution for modem answer containing no space after ":".
It is strange I can not submit the patch in a normal way (neither txt nor the more funny docx), here it is (i am afraid it will become distorted, so to get it as a patch you can write me at je_at_ktf.rtu.lv):

diff -rupN python-gsmmodem-0.9/gsmmodem/modem.py python-gsmmodem-0.9.1/gsmmodem/modem.py
--- python-gsmmodem-0.9/gsmmodem/modem.py	2013-07-18 12:59:16.000000000 +0300
+++ python-gsmmodem-0.9.1/gsmmodem/modem.py	2015-11-29 11:10:40.106263618 +0200
@@ -161,7 +161,7 @@ class GsmModem(SerialComms):
         self._smsMemWrite = None # Preferred message storage memory for writes (<mem2> parameter used for +CPMS)
         self._smsReadSupported = True # Whether or not reading SMS messages is supported via AT commands
 
-    def connect(self, pin=None):
+    def connect(self, pin=None, SMSC=None):
         """ Opens the port and initializes the modem and SIM card
          
         @param pin: The SIM card PIN code, if any
@@ -285,10 +285,12 @@ class GsmModem(SerialComms):
 
         # General meta-information setup
         self.write('AT+COPS=3,0', parseError=False) # Use long alphanumeric name format
-                
         # SMS setup
         self.write('AT+CMGF={0}'.format(1 if self._smsTextMode else 0)) # Switch to text or PDU mode for SMS messages
         self._compileSmsRegexes()
+        #setting SMSC from cmd-line argument
+        if SMSC != None:
+            self._smscNumber = SMSC
         if self._smscNumber != None:
             self.write('AT+CSCA="{0}"'.format(self._smscNumber)) # Set default SMSC number
             currentSmscNumber = self._smscNumber
@@ -301,7 +303,6 @@ class GsmModem(SerialComms):
         # ...check SMSC again to ensure it did not change
         if currentSmscNumber != None and self.smsc != currentSmscNumber:
             self.smsc = currentSmscNumber
-
         # Set message storage, but first check what the modem supports - example response: +CPMS: (("SM","BM","SR"),("SM"))
         try:
             cpmsLine = lineStartingWith('+CPMS', self.write('AT+CPMS=?'))
@@ -363,7 +364,9 @@ class GsmModem(SerialComms):
     def _unlockSim(self, pin):
         """ Unlocks the SIM card using the specified PIN (if necessary, else does nothing) """
         # Unlock the SIM card if needed
-        if self.write('AT+CPIN?')[0] != '+CPIN: READY':
+#DO NOT KNOW HOW TO STRIP SPACE FROM MODEM ANSWER AND HAVE NO POSSIBILTY TO TEST, SO I DELETED IT
+#        if self.write('AT+CPIN?')[0] != '+CPIN: READY':
+        if self.write('AT+CPIN?')[0] != '+CPIN:READY':
             if pin != None:
                 self.write('AT+CPIN="{0}"'.format(pin))
             else:
@@ -556,7 +559,6 @@ class GsmModem(SerialComms):
             if self.alive:
                 self.write('AT+CSCA="{0}"'.format(smscNumber))
             self._smscNumber = smscNumber
-
     def waitForNetworkCoverage(self, timeout=None):
         """ Block until the modem has GSM network coverage.
         
diff -rupN python-gsmmodem-0.9/python_gsmmodem.egg-info/requires.txt python-gsmmodem-0.9.1/python_gsmmodem.egg-info/requires.txt
--- python-gsmmodem-0.9/python_gsmmodem.egg-info/requires.txt	2013-07-18 14:21:47.000000000 +0300
+++ python-gsmmodem-0.9.1/python_gsmmodem.egg-info/requires.txt	2015-11-28 23:08:03.094136850 +0200
@@ -1 +1 @@
-pyserial>=2.6
\ No newline at end of file
+pyserial>=2.6
diff -rupN python-gsmmodem-0.9/python_gsmmodem.egg-info/top_level.txt python-gsmmodem-0.9.1/python_gsmmodem.egg-info/top_level.txt
--- python-gsmmodem-0.9/python_gsmmodem.egg-info/top_level.txt	2013-07-18 14:21:47.000000000 +0300
+++ python-gsmmodem-0.9.1/python_gsmmodem.egg-info/top_level.txt	2015-11-28 23:08:03.114135928 +0200
@@ -1,2 +1,2 @@
-gsmtermlib
 gsmmodem
+gsmtermlib
diff -rupN python-gsmmodem-0.9/tools/sendsms.py python-gsmmodem-0.9.1/tools/sendsms.py
--- python-gsmmodem-0.9/tools/sendsms.py	2013-07-18 13:32:32.000000000 +0300
+++ python-gsmmodem-0.9.1/tools/sendsms.py	2015-11-29 11:00:16.904992851 +0200
@@ -19,6 +19,7 @@ def parseArgs():
     parser.add_argument('-i', '--port', metavar='PORT', help='port to which the GSM modem is connected; a number or a device name.')
     parser.add_argument('-b', '--baud', metavar='BAUDRATE', default=115200, help='set baud rate')
     parser.add_argument('-p', '--pin', metavar='PIN', default=None, help='SIM card PIN')
+    parser.add_argument('-s', '--smsc',  metavar='SMSC', default=None, help='set specific SMSC number')
     parser.add_argument('-d', '--deliver',  action='store_true', help='wait for SMS delivery report')
     parser.add_argument('destination', metavar='DESTINATION', help='destination mobile number')    
     return parser.parse_args()
@@ -30,6 +31,7 @@ def parseArgsPy26():
     parser.add_option('-i', '--port', metavar='PORT', help='port to which the GSM modem is connected; a number or a device name.')
     parser.add_option('-b', '--baud', metavar='BAUDRATE', default=115200, help='set baud rate')
     parser.add_option('-p', '--pin', metavar='PIN', default=None, help='SIM card PIN')
+    parser.add_option('-s', '--smsc',  metavar='SMSC', default=None, help='set specific SMSC number')
     parser.add_option('-d', '--deliver',  action='store_true', help='wait for SMS delivery report')
     parser.add_positional_argument(Option('--destination', metavar='DESTINATION', help='destination mobile number'))    
     options, args = parser.parse_args()
@@ -50,7 +52,7 @@ def main():
     
     print('Connecting to GSM modem on {0}...'.format(args.port))
     try:
-        modem.connect(args.pin)
+        modem.connect(args.pin,args.smsc)
     except PinRequiredError:
         sys.stderr.write('Error: SIM card PIN required. Please specify a PIN with the -p argument.\n')
         sys.exit(1)

[Uploading smsc.docx…]()

mehdilauters, 2017-02-13:
+1 with a ````05c6:6000 Qualcomm, Inc. Siemens SG75````

tomchy, 2017-03-07:
What about:
```python
if (self.write('AT+CPIN?')[0]).replace(" ","") != '+CPIN:READY':
```
It is not a clean solution, but should not break unit tests for other modems :slightly_smiling_face: 
You may also include this fix inside a new [python-gsmmodem](http://github.com/babca/python-gsmmodem) version.