faucamp/python-gsmmodem #68: modem.connect(PIN) does not work in Python 3.x

deepaktalwardt, 2016-03-17:
I receive the following exception when using sms_handler_demo.py in python 3.4:

`. . .
. . .
File "/usr/local/lib/python3.4/dist-packages/python_gsmmodem-0.9-py3.4egg/gsmmodem/serial_comms.py", line 127, in write
File "/usr/lib/python3/dist-packages/serial/serialposix.py", line 475, in write
 n = os.write(self.fd, d)
TypeError: 'str' does not support the buffer interface`

A google search showed that this is because `string` needs to be converted to `bytes` before using the `write` function. Is there a work around for this? It is required for me to run the rest of my code in python 3.4. Any help would be appreciated. Thank you!

alex-eri, 2016-06-04:
Seems python3 is not supported here with new pyserial.
I tried to convert data, terminators in modem.py. No success. 

expectResponseTermSeq needs to be reimplemented in serial_comms.py

saviour123, 2017-11-06:
have you submitted a pull request because I have downgrade to python 2

alex-eri, 2017-12-26:
My port breaks python2 support.

I abandoned this port not completed.

Coding new library (redused number of functions) is easier than porting this.

saviour123, 2017-12-27:
I have to write my own toolkit around pyserial. I ported a pduconvertor to python3 as well. You have to convert the string to pdu before writing it to port<if u dealing with usb modem> https://github.com/saviour123/PDU-USSDConverter