faucamp/python-gsmmodem #77: CPIN TimeoutException

CyberBaby, 2016-09-02:
When I execute the example, I got the timeout exception. However, i am able to get the correct result when I issue the AT command in terminal. Any ideas?
----------------------------------------------------------------------------------------------------------
$ sms_handler_demo.py
Initializing modem...
INFO: Connecting to modem on port COM3 at 115200bps
DEBUG: write: ATZ
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['ATE0\r', 'OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
Traceback (most recent call last):
  File "D:\sms\sms_handler_demo.py", line 40, in <module>
    main()
  File "D:\sms\sms_handler_demo.py", line 32, in main
    modem.connect(PIN)
  File "C:\Python27\lib\site-packages\gsmmodem\modem.py", line 198, in connect
    self._unlockSim(pin)
  File "C:\Python27\lib\site-packages\gsmmodem\modem.py", line 366, in _unlockSim
    if self.write('AT+CPIN?')[0] != '+CPIN: READY':
  File "C:\Python27\lib\site-packages\gsmmodem\modem.py", line 398, in write
    responseLines = SerialComms.write(self, data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "C:\Python27\lib\site-packages\gsmmodem\serial_comms.py", line 135, in write
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException

CyberBaby, 2016-09-04:
Finally, it works after I commented the "unlock" call.

leoganda, 2016-09-10:
@CyberBaby , I am experiencing the same issue. Which "unlock" call you commented to made it works?

babca, 2016-10-25:
Duplicate issue https://github.com/faucamp/python-gsmmodem/issues/50. Now resolved.

The timeout was too low.
See my fork with all 2014-2016 fixes included: https://github.com/babca/python-gsmmodem/issues/11