faucamp/python-gsmmodem #81: PDU creates invalid messages when > allowed length

BMtthws, 2017-03-07:
The pdu.py splits long messages up depending on the type of string (GSM-7, UCS2,8-bit)
But the calculated lengths are wrong and cause modem errors.

attached is a fixed version that works with Multitech MTD-H5
[pdu.py.txt](https://github.com/faucamp/python-gsmmodem/files/825914/pdu.py.txt)

tomchy, 2017-03-07:
Hi!
could you test if this issue occurs inside following [updated fork](http://github.com/babca/python-gsmmodem) ?
It has several updates and fixes, including long PDUs :slightly_smiling_face:.

BMtthws, 2017-03-07:
chunks calculation in divideTextUcs2 should be:

   fullChunksCount = int((len(plainText) -1) / MAX_MULTIPART_MESSAGE_LENGTH[0x08])

tomchy, 2017-03-08:
Could you explain that/provide an example?

Current implementation passes following unit tests (last test inside ``test_pdu.py``):
```python
    def test_encode_Ucs2_divideSMS(self):
        """ Tests whether text will be devided into a correct number of chunks while using UCS-2 alphabet"""
        text = "12345-010 12345-020 12345-030 12345-040 12345-050 12345-060"
        self.assertEqual(len(gsmmodem.pdu.divideTextUcs2(text)), 1)
        text = "12345-010 12345-020 12345-030 12345-040 12345-050 12345-060 1234567"
        self.assertEqual(len(gsmmodem.pdu.divideTextUcs2(text)), 1)
        text = "12345-010 12345-020 12345-030 12345-040 12345-050 12345-060 12345678"
        self.assertEqual(len(gsmmodem.pdu.divideTextUcs2(text)), 2)
        text = "12345-010 12345-020 12345-030 12345-040 12345-050 12345-060 123456["
        self.assertEqual(len(gsmmodem.pdu.divideTextUcs2(text)), 1)
        text = "12345-010 12345-020 12345-030 12345-040 12345-050 12345-060 1234567["
        self.assertEqual(len(gsmmodem.pdu.divideTextUcs2(text)), 2)
        text = "12345-010,12345-020,12345-030,12345-040,12345-050,12345-060,123456 12345-010,12345-020,12345-030,12345-040,12345-050,12345-060,1234567"
        self.assertEqual(len(gsmmodem.pdu.divideTextUcs2(text)), 2)
```

BMtthws, 2017-03-08:
Hi,

     I think it is OK as is.

     if a ucs-2 buffer is exactly 134 bytes long, your calc results in 2 fullChunks and no remainder, whereas my pedantically correct way results in 1 fullChunk and 67 remainder.

   Either way it works.

    I will try your code tomorrow. It looks OK to me, as it handles the shortened buffer size it needs to  once it is chunked.

Brian