faucamp/python-gsmmodem #82: please do not enable delivery report by default

sherpya, 2017-04-04:
the line
`self.write('AT+CSMP=49,167,0,0', parseError=False) # Enable delivery reports`

enables delivery reports that have often a cost

If I'm correct it should be
`self.write('AT+CSMP=33,167,0,0', parseError=False) # Enable delivery reports`

anyway it would be better to have it configurable

sherpya, 2017-04-04:
I've also noticed encodeSmsSubmitPdu has a default keyword arg `requestStatusReport=True`

tomchy, 2017-04-19:
Please switch to the [updated fork](http://github.com/babca/python-gsmmodem) :slightly_smiling_face: 
There is optional parameter "requestDelivery=True" inside GsmModem class which sets the option that you have mentioned:
```python
self.write('AT+CSMP=17,167,0,0', parseError=False) # Not enable delivery reports
```