faucamp/python-gsmmodem #85: Timeout making voice call, Huawei e171

hobodrifterdavid, 2017-06-16:
I tried running the test script dial_callback_demo.py. Ubuntu 16.04, Huawei e171 USB 3G modem (unlocked and voice enabled). I get this message:

DEBUG: write: AT+CSQ
DEBUG: response: ['+CSQ: 10,99', 'OK']
Dialing number: 0951161251
DEBUG: write: ATD0951161251;
DEBUG: response: ['OK']
Traceback (most recent call last):
  File "dial_callback_demo.py", line 71, in <module>
    main()
  File "dial_callback_demo.py", line 64, in main
    call = modem.dial(NUMBER, timeout=10, callStatusUpdateCallbackFunc=callStatusCallback)
  File "/usr/local/lib/python2.7/dist-packages/gsmmodem/modem.py", line 728, in dial
    raise TimeoutException()
gsmmodem.exceptions.TimeoutException

I plan to look into it myself, but, any hints?

Many Thanks, David

vishalashapur, 2017-10-10:
Hi i am also facing the same problem. below is my log. can i get any hint to resolve this..!
Initializing modem...
Waiting for network coverage...
Dialing number: +918660701459
Traceback (most recent call last):
  File "dial_polling_demo.py", line 69, in <module>
    main()
  File "dial_polling_demo.py", line 36, in main
    call = modem.dial(NUMBER)
  File "build/bdist.linux-x86_64/egg/gsmmodem/modem.py", line 742, in dial
gsmmodem.exceptions.TimeoutException: None

TAdegbola, 2017-10-18:
I have a similar problem with the Huawei modem E3531.  I tried  incoming_callTrial.py and it got stuck at the ¨Waiting for incoming calls.... ¨ message.  I called the modem with my phone and I get a line busy message on my phone.
I then tried the dial_demo.py code and the following is what I got.  Any ideas of what I can do to get me out of this bind?

Initializing modem...
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: ATZ
DEBUG: notification: ['^RSSI: 13', '^HCSQ:"WCDMA",49,35,37', '^RSSI: 13', '^HCSQ:"WCDMA",50,34,33', '^RSSI: 13', '^HCSQ:"WCDMA",50,35,35']
DEBUG: Unhandled unsolicited modem notification: ['^RSSI: 13', '^HCSQ:"WCDMA",49,35,37', '^RSSI: 13', '^HCSQ:"WCDMA",50,34,33', '^RSSI: 13', '^HCSQ:"WCDMA",50,35,35']
DEBUG: response: ['OK']
DEBUG: write: ATE0
DEBUG: response: ['OK']
DEBUG: write: AT+CFUN?
DEBUG: response: ['+CFUN: 1', 'OK']
DEBUG: write: AT+CMEE=1
DEBUG: response: ['OK']
DEBUG: write: AT+CPIN?
DEBUG: response: ['+CPIN: READY', 'OK']
DEBUG: write: AT+CLAC
DEBUG: response: ['ERROR']
DEBUG: write: AT+WIND?
DEBUG: response: ['ERROR']
DEBUG: write: AT+CGMI
DEBUG: response: ['huawei', 'OK']
INFO: Loading Huawei call state update table
DEBUG: write: AT+COPS=3,0
DEBUG: response: ['OK']
DEBUG: write: AT+CMGF=0
DEBUG: response: ['OK']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+2348020000008",145', 'OK']
DEBUG: write: AT+CSMP=49,167,0,0
DEBUG: response: ['ERROR']
DEBUG: write: AT+CSCA?
DEBUG: response: ['+CSCA: "+2348020000008",145', 'OK']
DEBUG: write: AT+CPMS=?
DEBUG: response: ['+CPMS: ("SM","ME"),("SM","ME"),("SM","ME")', 'OK']
DEBUG: write: AT+CPMS="ME","ME","ME"
DEBUG: response: ['+CPMS: 0,20,0,20,0,20', 'OK']
DEBUG: write: AT+CNMI=2,1,0,2
DEBUG: response: ['OK']
DEBUG: write: AT+CLIP=1
DEBUG: response: ['OK']
DEBUG: write: AT+CRC=1
DEBUG: response: ['OK']
DEBUG: write: AT+CVHU=0
DEBUG: response: ['ERROR']
Waiting for network coverage...
DEBUG: write: AT+CREG?
DEBUG: response: ['+CREG: 2,1,"018B","002D9CE7"', 'OK']
INFO: +CREG check disabled due to invalid response or unsupported command
DEBUG: write: AT+CSQ
DEBUG: response: ['+CSQ: 13,99', 'OK']
Dialing number: 08034019398
DEBUG: write: ATD08034019398;

Traceback (most recent call last):
  File "/home/pi/GSMTest/python-gsmmodem/CallTest.py", line 61, in <module>
    main()
  File "/home/pi/GSMTest/python-gsmmodem/CallTest.py", line 46, in main
    call = modem.dial(NUMBER)
  File "/home/pi/GSMTest/python-gsmmodem/gsmmodem/modem.py", line 716, in dial
    self.write('ATD{0};'.format(number), timeout=timeout, waitForResponse=self._waitForAtdResponse)
  File "/home/pi/GSMTest/python-gsmmodem/gsmmodem/modem.py", line 413, in write
    responseLines = super(GsmModem, self).write(data + writeTerm, waitForResponse=waitForResponse, timeout=timeout, expectedResponseTermSeq=expectedResponseTermSeq)
  File "/home/pi/GSMTest/python-gsmmodem/gsmmodem/serial_comms.py", line 138, in write
    raise TimeoutException(self._response)
TimeoutException: ['NO CARRIER', '^RSSI: 13', '^HCSQ:"WCDMA",50,34,33']

TAdegbola, 2017-10-18:
By the way, the SMS_handler_demo.py worked.  I was able to send an SMS to my Raspberry Pi 3 and I got a reply.