faucamp/python-gsmmodem #92: How do you send accented characters?

silgon, 2017-11-15:
I would like to know how to send strings with accented characters. I don't find how (maybe I haven't look enough tough). So, let's say just to test, we have:

     str_to_send="áóéú"

tomchy, 2017-11-29:
It should be done the same way as message with ASCII characters. There are some procedures that will try to encode it with GSM-7 character set, fail and send it using UTF-8. 

If you are sending it from python script, check if setting character set inside it to UTF-8 solves your problem 🙂

silgon, 2018-09-21:
Sorry for the super late answer. I needed that again today. You're completely right. Also tested it with python 3 and it works perfectly fine with `python-gsmmodem-new`.

closed 2018-09-21 by silgon