faucamp/python-gsmmodem #94: TypeError: unicode strings are not supported, please encode to bytes: 'ATZ\r'

filozof71, 2018-04-26:
Hello,

I quite new to Python, I am playing with GSM mode for two weak. I hvae the same error in windows nad Ubuntu.

If I tried to code some AT commands myself I can use encode() method but is there any setting so that this library could work properly?

###############
Python 3.5.2 (default, Nov 23 2017, 16:37:01) 
[GCC 5.4.0 20160609] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import logging
>>> PORT = '/dev/ttyUSB0'
>>> BAUDRATE = 115200
>>> logging.basicConfig(format='%(levelname)s: %(message)s', level=logging.DEBUG)
>>> from gsmmodem.modem import GsmModem
>>> modem = GsmModem(PORT, BAUDRATE)
>>> modem.smsTextMode = True
>>> modem.connect()
INFO: Connecting to modem on port /dev/ttyUSB0 at 115200bps
DEBUG: write: ATZ
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/usr/local/lib/python3.5/dist-packages/python_gsmmodem-0.9-py3.5.egg/gsmmodem/modem.py", line 177, in connect
  File "/usr/local/lib/python3.5/dist-packages/python_gsmmodem-0.9-py3.5.egg/gsmmodem/modem.py", line 413, in write
  File "/usr/local/lib/python3.5/dist-packages/python_gsmmodem-0.9-py3.5.egg/gsmmodem/serial_comms.py", line 127, in write
  File "/home/fil71/.local/lib/python3.5/site-packages/serial/serialposix.py", line 532, in write
    d = to_bytes(data)
  File "/home/fil71/.local/lib/python3.5/site-packages/serial/serialutil.py", line 63, in to_bytes
    raise TypeError('unicode strings are not supported, please encode to bytes: {!r}'.format(seq))
TypeError: unicode strings are not supported, please encode to bytes: 'ATZ\r'

tomchy, 2018-04-26:
Python3 support in this library was broken. Please try python-gsmmodem-new, which is a forked and still maintained version of this library.

filozof71, 2018-04-26:
@tomchy Thank You very much. Indeed python-gsmmodem-new does not have this problem, so I cam close this issue.

closed 2018-04-26 by filozof71