faucamp/python-gsmmodem #95: get error while installing

ghorbanpirizad, 2018-05-08:
I get this error when want install in both py2 , py3 why.

tomchy, 2018-05-08:
Hi!
It seems that your problen is not related to the library itself, but to the python interpreter that you are using ( [similar issue](https://stackoverflow.com/questions/49822079/getting-docstring-in-qpython3) ).

Additionally you may consider moving to the python-gsmmodem-new package, which is recommended if you want to use it under python3 or with non-ascii characters.

ghorbanpirizad, 2018-05-08:
I get same issue on python-gsmmodem-new. 
Is this for my py ver. That's 3.2 or because android?!

tomchy, 2018-08-16:
I have a strong feeling that it is because you are using QPython. You may also modify manually the package and replace `__doc__` with a meaningful text, but this change will not be up-merged as the usage of `__doc__` is 100% valid in python.