faucamp/python-gsmmodem #99: does it work on Raspberry Pi?

ashfaqahmed, 2018-07-22:
does it work on Raspberry Pi? as i have been facing timeout error on correct port, any suggestions or anyone has tested this on it?

LeVadim, 2018-08-02:
Same issue here, thanks

tomchy, 2018-08-02:
0. Please try a non-abandoned fork of this repository:
https://github.com/babca/python-gsmmodem

1. What type od modem do you use? 

2. Have you checked all serial ports that the device registers?

3. Are you able to connect to the modem manually (e.g. by picocom or screen)? The device should reply with `OK` after sending `AT` command. Be careful about newline characters.

QuentinRasidy, 2020-07-02:
Hi everyone, 

I have it working on Raspberry Pi. 
Although I needed to use a USB hub (with power) because the Raspberry Pi itself does not provide enough power for the model to work. I think this is a common issue with USB modems and Raspberry Pies.

Maybe you are running into the same issue ? 

Quentin