faucamp/python-gsmmodem #104: Read SMS without storage

seeo2, 2020-04-13:
This Python libary is fantastic!  Fall in love with Python partly because of this.  Made my life much easier.

I'm developing a SMS receiver which constantly receive SMS at 1-minute interval.  Looking into the code it used ME storage.  I'm just afraid the storage hardware may fail in a few months time due to wear and tear.  Instead, it is possible to receive SMS without storage?

Sorry if this is wrong place as I'm new to Github too.

tomchy, 2020-05-06:
Hello!
There is a `delete` parameter inside `listStoredSms` modem's API that will issue SMS deletion right after reading them from modem's memory

Additionally - if you find any issues, please check the [updated fork](https://github.com/babca/python-gsmmodem/) as this repository is no longer maintained.

seeo2, 2020-05-19:
Appreciate that, Tomchy!

I figured out a way to bypass any memory storage by setting CNMI=1,2,0,0,0.  SMS will send directly to TE.  In this way, memory will not be wear and tear even for bulk SMS reception.